platformio/platformio-examples #1: Error: No such command "test"

jantimon, 2016-06-26:
```bash 
platformio --version
PlatformIO, version 2.10.4.dev2

platformio test
Usage: platformio [OPTIONS] COMMAND [ARGS]...

Error: No such command "test"
```

jantimon, 2016-06-26:
Oh you need platformio 3

Please add a note to the readme
```
pip install https://github.com/platformio/platformio/archive/feature/platformio-30.zip
```

ivankravets, 2016-06-26:
@jantimon here did you find this example? It is not located in the `master` or even in the `develop` branch. It is located in a `feature`. 

PlatformIO 3.0 has not been released yet and is still in development. Please follow us on Twitter [@PlatformIO_Org](https://twitter.com/PlatformIO_Org) and we will inform about the first beta.

Thanks.

jantimon, 2016-06-26:
Googled for `"platformio" tests` first result: http://docs.platformio.org/en/feature-platformio-30/userguide/cmd_test.html
Looks exactly like the normal platformio page.. didn't realise the small text in the bottom says 3.0

jantimon, 2016-06-26:
Was really disappointed by platformio because it says it supports "Continuous Integration" however there is no way to get unit tests running with travis or on the local machine. so the main part of ci which would allow working on software in teams is not included.

ivankravets, 2016-06-26:
> Was really disappointed by platformio because it says it supports "Continuous Integration" however there is no way to get unit tests running with travis or on the local machine

You mix 2 different things:

1. Continuous Integration
2. Hardware Unit Testing

The 1-st item is implemented in PlatformIO 2.0 and a lot of developers/companies use it. The docs: http://docs.platformio.org/en/latest/ci/travis.html

The 2-nd item means that tests will be run on TARGET device, not on Travis. It requires hardware device. 

Nevertheless, we opened issue to allow run UNIT tests on the local machine: https://github.com/platformio/platformio/issues/519 . It is not equal with `platformio ci` command and should run unit tests using Unit Test Framework dicrectly on the PC.

If you need any help with CI, I'm glad to help you.

ivankravets, 2016-10-02:
Unit Testing has been introduced in PlatformIO 3.0 http://docs.platformio.org/en/stable/plus/unit-testing.html

closed 2016-10-02 by ivankravets

jantimon, 2016-10-03:
Wow cool - this is exactly what I was looking for! :) 👍