platformio/platformio-examples #4: *** [upload] /dev/cu.usbmodem1413/firmware.bin: Not a directory

pdemil, 2016-11-26:
Trying to run an example (mbed-blink) on the nucleo_l152_re platform.

Troubles with upload-port...

platformio device list
/dev/cu.usbmodem1413
--------------------
Hardware ID: USB VID:PID=0483:374B SER=066AFF565554836767235840 LOCATION=20-1
Description: STM32 STLink

sudo platformio run -e nucleo_l152re --target upload --upload-port /dev/cu.usbmodem1413

[Sat Nov 26 10:26:50 2016] Processing nucleo_l152re (platform: ststm32, board: nucleo_l152re, framework: mbed)
----------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 7 compatible libraries
Looking for dependencies...
Project does not have dependencies
Looking for upload disk...
Use manually specified: /dev/cu.usbmodem1413
Uploading .pioenvs/nucleo_l152re/firmware.bin
*** [upload] /dev/cu.usbmodem1413/firmware.bin: Not a directory
================================================================== [ERROR] Took 1.16 seconds ==================================================================

ivankravets, 2016-11-26:
Does this board have ARM MBED media disk? If yes, please specify upload port as path to this disk or live it empty. PlatformIO will automatically detect it. 

Otherwise, set `upload_protocol = stlink` in configuration file.

pdemil, 2016-11-26:
Fast response, great!

I've added the upload_protocol = stlink

sudo platformio run -e nucleo_l152re --target upload 

[Sat Nov 26 10:46:21 2016] Processing nucleo_l152re (upload_protocol: stlink, platform: ststm32, board: nucleo_l152re, framework: mbed)
----------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 7 compatible libraries
Looking for dependencies...
Project does not have dependencies
Uploading .pioenvs/nucleo_l152re/firmware.bin
2016-11-26T10:46:25 INFO src/stlink-common.c: Loading device parameters....
2016-11-26T10:46:25 WARN src/stlink-common.c: unknown chip id! 0xe0042000
[!] send_recv
*** [upload] Error 255
=======================

Good news is that it is trying to do something..

I see a fail.txt on the nucleo media disk, SWD_ERROR. 
Maybe that is not related to platformio, trying to run it on Mac OS X 10.11.3

ivankravets, 2016-11-26:
Try to comment upload protocol if you board has mbed bootloader with mbed disk

pdemil, 2016-11-26:
Maybe related to https://github.com/platformio/platformio/issues/819 ?

ivankravets, 2016-11-26:
Please try development version http://docs.platformio.org/en/latest/installation.html#development-version

pdemil, 2016-11-26:
Using development version now,

_when trying with st-link:_

st-flash write .pioenvs/nucleo_l152re/firmware.bin 0x08000000
2016-11-26T14:09:36 INFO src/stlink-common.c: Loading device parameters....
2016-11-26T14:09:36 INFO src/stlink-common.c: Device connected is: L152RE, id 0x10006437
2016-11-26T14:09:36 INFO src/stlink-common.c: SRAM size: 0x14000 bytes (80 KiB), Flash: 0x80000 bytes (512 KiB) in pages of 256 bytes
2016-11-26T14:09:36 INFO src/stlink-common.c: Ignoring 4 bytes of 0x00 at end of file
2016-11-26T14:09:36 INFO src/stlink-common.c: Attempting to write 33132 (0x816c) bytes to stm32 address: 134217728 (0x8000000)
[!] send_recv
2016-11-26T14:09:37 INFO src/stlink-common.c: Finished erasing 130 pages of 256 (0x100) bytes
Flash page at addr: 0x08006b002016-11-26T14:09:37 INFO src/stlink-common.c: Starting Half page flash write for STM32L core id
Flash page at addr: 0x08008100 erased
Error: Data length doesn't have a 32 bit alignment: +2 byte.
*** [upload] Error -6

_when trying with mbed disk_
sudo platformio run -e nucleo_l152re --target upload --upload-port /VOLUMES/NODE_L152RE/ -v

[Sat Nov 26 14:09:02 2016] Processing nucleo_l152re (platform: ststm32, board: nucleo_l152re, framework: mbed)

Collected 7 compatible libraries
Looking for dependencies...
Project does not have dependencies
MethodWrapper(["upload"], [".pioenvs/nucleo_l152re/firmware.bin"])
Use manually specified: /VOLUMES/NODE_L152RE/
MethodWrapper(["upload"], [".pioenvs/nucleo_l152re/firmware.bin"])
*** [upload] No space left on device

ivankravets, 2016-11-26:
> when trying with st-link:

Please file issue here https://github.com/platformio/platform-ststm32/issues

> *** [upload] No space left on device

Issue https://github.com/platformio/platformio/issues/669

Try manually copy .pioenvs/nucleo_l152re/firmware.bin to MBED disk.

closed 2016-12-02 by ivankravets