platformio/platformio-examples #12: Problem with platform.io and ArduinoOTA (Example BasicOTA.ino)

chessweb01, 2017-03-03:
The example BasicOTA.ino that illustrates over the air firmware updates of ESP8266 modules compiles with Arduino 1.6.8 but not with Atom/platform.io. 

With the latter I get "error: 'class ArduinoOTAClass' has no member named 'getCommand'". platform.io somehow incorporates an obsolete version of the ArduinoOTA class.

I'm positive that Atom, platform.io, and all packages are up to date on my computer.

ivankravets, 2017-03-04:
Have you resolved this issue?

scropion86, 2017-05-09:
i have the same issue , even after applying the workaround still not compiling
and below error is appearing

```

pio
lib
ArduinoOTA.cpp
ArduinoOTA.h
readme.txt
src
main.cpp
.gitignore
.travis.yml
platformio.ini
main.cpp
platformio.ini
 1
 1
 2
 3
 4
 5
 6
 7
 8
 9
10
11
12
13
14
15

; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html
 
[env:esp12e]
platform = espressif8266
board = esp12e
framework = arduino
 
2.0 s
platformio run
src\main.cpp:35:20: error: 'class ArduinoOTAClass' has no member named 'getCommand'
if (ArduinoOTA.getCommand() == U_FLASH)
^
Compiling .pioenvs\esp12e\FrameworkArduino\base64.o
*** [.pioenvs\esp12e\src\main.o] Error 1
 [ERROR] Took 1.58 seconds
 
```

ivankravets, 2018-02-03:
Please open new issue here https://github.com/platformio/platform-espressif8266 if you still experience this issue.

closed 2018-02-03 by ivankravets