platformio/platformio-examples #13: Providing examples for OTA firmware updates without authentication is dangerous and irresponsible.

dlmetcalf, 2017-04-10:
Looking at this (https://github.com/platformio/platformio-examples/tree/develop/espressif/esp8266-arduino-ota), there appears to be absolutely no password or authentication required to remotely reflash the device at all!

This is extremely irresponsible and the type of thing that leads to criminal botnets etc.  Many developers *will* copy and paste the code as is.  Safe use of the function should be demonstrated (including unique, non-default passwords), or no demonstration should be provided at all.

Note even the upstream OTA library setPassword implementation uses an unsalted, single pass of the vulnerable and deprecated MD5 algorithm to store passwords, rather than a password hashing function.  Amongst other issues (including a repeating, timer based nonce).  But at least is very slightly better than having absolutely no security at all.  

Please find and use a suitable implementation. (I'd recommend https://github.com/jedisct1/libhydrogen/, but it's still a work in progress).

To learn more about IoT security, please see the following resources:

Hinden, B., March 2017, "The Internet of Insecure Things", The Internet Protocol Journal, Vol.20, No. 1, p.12, ISSN 1944-1134, accessed 10 April 2017, http://ipj.dreamhosters.com/wp-content/uploads/issues/2017/ipj20-1.pdf

IETF Internet Architecture Board, 2017, "Report from the Internet of Things (IoT) Software Update (IoTSU) Workshop 2016", draft-iab-iotsu-workshop-01, https://datatracker.ietf.org/doc/draft-iab-iotsu-workshop/?include_text=1

IoT Security Foundation, 2016, “IoT Security Compliance Framework”, Release 1, accessed 9 April 2017, https://iotsecurityfoundation.org/wp-content/uploads/2016/12/IoT-Security-Compliance-Framework.pdf

Cloud Security Alliance, 2016, “Future-proofing the Connected World: 13 Steps to Developing Secure IoT Products”, accessed 9 April 2017, https://downloads.cloudsecurityalliance.org/assets/research/internet-of-things/future-proofing-the-connected-world.pdf

U.S. Department of Homeland Security, 2016, "Strategic Principles for Securing the Internet of Things (IoT)", Version 1.0,  https://www.dhs.gov/sites/default/files/publications/Strategic_Principles_for_Securing_the_Internet_of_Things-2016-1115-FINAL_v2-dg11.pdf

U.S. National Institute of Standards and Technology, 2014, "Framework for Improving Critical Infrastructure Cybersecurity", Version 1.0,  https://www.nist.gov/sites/default/files/documents/cyberframework/cybersecurity-framework-021214.pdf

Industrial Internet Consortium, 2016, “Industrial Internet of Things Volume G4: Security Framework”, accessed 9 April 2017, http://www.iiconsortium.org/pdf/IIC_PUB_G4_V1.00_PB-3.pdf

Online Trust Alliance, 2017, “IoT Trust Framework”, accessed 9 April 2017, http://otalliance.actonsoftware.com/acton/attachment/6361/f-008d/1/-/-/-/-/IoT%20Trust%20Framework.pdf

To learn more about WHY you should securing systems, see examples like these:

Krebs, B., 2012, “The Scrap Value of a Hacked PC, Revisited”, Krebs On Security blog, accessed 9 April 2017, <https://krebsonsecurity.com/2012/10/the-scrap-value-of-a-hacked-pc-revisited/>

Neumann, P., 2015, Illustrative Risks to the Public in the Use of Computer Systems and Related Technology, Computer Science Laboratory, SRI International, Menlo Park CA, accessed 9 April 2017, <http://www.csl.sri.com/users/neumann/illustrative.html>

https://en.wikipedia.org/wiki/2016_Dyn_cyberattack#Affected_services

vayulove, 2017-11-13:
These are example programs and a gift to the community. You're believing someone giving something for free has a responsibility. The responsibility for what a user or developer does or doesn't do rests with them. I'm thankful that someone takes the time to give me something and I don't expect them to give it to me on a silver platter.   I would regard your post as helpful if the tone of it were here's my gift of security practices to add to these examples, but as it stands the tone is of blame and demanding responsibility.

ivankravets, 2018-02-03:
We removed that example. Thanks!

closed 2018-02-03 by ivankravets