platformio/platformio-examples #18: ** [upload] /dev/ttyACM0/firmware.bin: Not a directory

khoa-io, 2017-08-17:
Hi all.
I'm trying to run an example on a Freescale board. Building is OK but I can't upload file to the board.
Here is my upload command and its output:
```
pio run --target upload --upload-port /dev/ttyACM0 -v
[Thu Aug 17 20:47:00 2017] Processing frdm_kl46z (platform: freescalekinetis; upload_port: /dev/ttyACM0; board: frdm_kl46z; framework: mbed)
---------------------------------------------------------------------------------------------------------
Collected 22 compatible libraries
Looking for dependencies...
Project does not have dependencies
MethodWrapper(["upload"], [".pioenvs/frdm_kl46z/firmware.bin"])
Use manually specified: /dev/ttyACM0
MethodWrapper(["upload"], [".pioenvs/frdm_kl46z/firmware.bin"])
*** [upload] /dev/ttyACM0/firmware.bin: Not a directory
======================================= [ERROR] Took 1.32 seconds =======================================
```
I searched a lot on Google but there's no information can help me. So I create an issue here.
Thank you!

ivankravets, 2017-08-17:
Please move issue to https://github.com/platformio/platform-freescalekinetis

closed 2017-08-17 by ivankravets

khoa-io, 2017-08-17:
Oh sorry. I do it now.