platformio/platformio-examples #21: Can't compile LCD Example using Platform IO, Blink and Hello World work just fine.

ilovetogetspamed, 2017-10-20:
Opened on Simba issues also https://github.com/eerimoq/simba/issues/161

# Error 
```
...
Compiling .pioenvs/megaatmega2560/src/main.o

src/main.c: In function 'main':
src/main.c:36:29: error: storage size of 'lcd' isn't known
struct hd44780_driver_t lcd;
...
```
# Source
```
#define PORT_HAS_HD44780   // I hoped adding this would solve the compilation error...
                                                      // see source for "simba.h"

#include "simba.h"

int main()
{
    struct hd44780_driver_t lcd;

    sys_start();

    hd44780_init(&lcd,
                 &pin_d8_dev,   // RS
                 &pin_d9_dev,   // ENABLE
                 &pin_d5_dev,   // DATA 4
                 &pin_d6_dev,   // DATA 5
                 &pin_d7_dev,   // DATA 6
                 &pin_d8_dev,   // DATA 7
                 2,             // ROWS
                 16);           // COLUMNS
    hd44780_start(&lcd);

    hd44780_display(&lcd, "Welcome!\r\n");
    hd44780_cursor_show(&lcd);

    while (1) {
        hd44780_put(&lcd, chan_getc(sys_get_stdin()));
    }

My LCD is wired a little differently (pin assignments are different) than the Simba example but it should still compile I would think.

ilovetogetspamed, 2017-10-20:
Darwin ********.*****.net 15.6.0 Darwin Kernel Version 15.6.0: Sun Jun  4 21:43:07 PDT 2017; root:xnu-3248.70.3~1/RELEASE_X86_64 x86_64

PlatformIO, version 3.5.0a16
Atmel AVR 1.71

```
[Fri Oct 20 18:19:24 2017] Processing megaatmega2560 (platform: atmelavr; board: megaatmega2560; framework: simba)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 0 compatible libraries
Looking for dependencies...
No dependencies
Compiling .pioenvs/megaatmega2560/src/main.o

src/main.c: In function 'main':
src/main.c:36:29: error: storage size of 'lcd' isn't known
struct hd44780_driver_t lcd;
^
src/main.c:40:5: warning: implicit declaration of function 'hd44780_init' [-Wimplicit-function-declaration]
hd44780_init(&lcd,
^
src/main.c:49:5: warning: implicit declaration of function 'hd44780_start' [-Wimplicit-function-declaration]
hd44780_start(&lcd);
^
src/main.c:51:5: warning: implicit declaration of function 'hd44780_display' [-Wimplicit-function-declaration]
hd44780_display(&lcd, "Welcome!\r\n");
^
src/main.c:52:5: warning: implicit declaration of function 'hd44780_cursor_show' [-Wimplicit-function-declaration]
hd44780_cursor_show(&lcd);
^
src/main.c:55:9: warning: implicit declaration of function 'hd44780_put' [-Wimplicit-function-declaration]
hd44780_put(&lcd, chan_getc(sys_get_stdin()));
^
src/main.c:55:9: warning: implicit declaration of function 'chan_getc' [-Wimplicit-function-declaration]
src/main.c:36:29: warning: unused variable 'lcd' [-Wunused-variable]
struct hd44780_driver_t lcd;
^
Compiling .pioenvs/megaatmega2560/SimbaFramework/3pp/compat/arch/sys_arch.o
Compiling .pioenvs/megaatmega2560/SimbaFramework/3pp/lwip-1.4.1/src/api/tcpip.o
*** [.pioenvs/megaatmega2560/src/main.o] Error 1
Compiling .pioenvs/megaatmega2560/SimbaFramework/3pp/lwip-1.4.1/src/core/def.o
========================== [ERROR] Took 0.96 seconds ==========================
 ```

ivankravets, 2018-02-03:
Please reopen this issue here https://github.com/platformio/platform-atmelavr/issues if you still have this problem.

closed 2018-02-03 by ivankravets