platformio/platformio-examples #24: Unit testing examples conveniently omit the fact that IT COSTS MONEY TO UNIT TEST

ifreecarve, 2018-01-08:
After seeing [Travis CI docs that recommend PlatformIO for CI testing](https://docs.travis-ci.com/user/integration/platformio/) I put some effort into adapting [the wiring-blink PlatformIO unit test example](https://github.com/platformio/platformio-examples/tree/develop/unit-testing/wiring-blink) into [my own sample project](https://github.com/ifreecarve/arduino-ci-unit-tests).  

After a few hours of work, and feeling pretty good about the progress, I ran `platformio test -e uno` and got this message:

```
$ platformio test -e uno
PlatformIO Plus (https://pioplus.com) v0.13.1
You are not logged in. Please log in to PIO Account using
`pio account login` command and try again.

If you don't have PIO Account yet, please create it using
`pio account register` command.
```

It's not unreasonable that you operate a service and [charge money for it](http://platformio.org/pricing).  However, it's inconsiderate that [your documentation](https://github.com/platformio/platformio-examples/blob/develop/unit-testing/wiring-blink/README.rst) makes absolutely no mention of it.

ivankravets, 2018-01-10:
We have. See http://docs.platformio.org/en/latest/plus/unit-testing.html

> New in version 3.0: (PlatformIO Plus)

Please click on "PlatformIO Plus"

ifreecarve, 2018-01-10:
Note that this bug report doesn't say "you didn't document it"; it says "you omitted it" from [the example](https://github.com/platformio/platformio-examples/blob/develop/unit-testing/wiring-blink/README.rst) and [the unit testing page itself](http://docs.platformio.org/en/latest/plus/unit-testing.html).  As a new user, it was not clear to me until I ran the unit test command that a login was required.

Based on your instruction to "Please click on "PlatformIO Plus"", you are assuming that (as a new user) I would know the following in advance: 

1. "New in version 3.0: (PlatformIO Plus)" is not a link to a changelog (indicating the version after which the feature is available) but instead a link to a pricing page **that is a prerequisite** for setting up unit testing.
2. That I will need to scroll down the page to look for "PIO Unit testing", which appears below the fold
3. That because unit testing appears under the paid plan listing, it is therefore not available as part of the free plan.  Or that "plus" is meant in the "you pay for it" sense, not the "C++" sense.

Expected unit testing documetation
-----------

> Unit testing is available in the [paid plans of PlatformIO Plus](http://platformio.org/pricing), as of version 3.0.  **Account setup is a prerequisite for unit testing.**

Actual unit testing documentation
----------

> New in version 3.0: ([PlatformIO Plus](http://platformio.org/pricing))

Expected example documentation
------------

1. Create a [PlatformIO account](http://docs.platformio.org/en/latest/plus/pio-account.html) and sign up for a [paid plan](http://platformio.org/pricing) that includes PIO Unit testing
2. Install [PlatformIO Core](http://docs.platformio.org/page/core.html)
3. Download [examples source code](https://github.com/platformio/platformio-examples/archive/develop.zip)
4. Extract ZIP archive
5. Run these commands:
```
 <the command to log in to PlatformIO, and/or check whether you're logged in>
```

Actual example Documentation
------------

1. Install [PlatformIO Core](http://docs.platformio.org/page/core.html)
2. Download [examples source code](https://github.com/platformio/platformio-examples/archive/develop.zip)
3. Extract ZIP archive
4. Run these commands:

rjopson, 2018-10-09:
I second this. I've spent about an hour getting everything set up only to realize it's paid. Not unreasonable, but would have been nice to know before I put the time in...

sleeper, 2018-12-19:
Great! Actually In understood my problem thanks to this issue ;) Ok, so off from Platformio: paying to be able to do unit tests (especially the native ones) is a no go to.

ivankravets, 2018-12-20:
Sorry, we are working on resolving this issue. Mostly, people pay for PIO Plus just to help us maintain everything that we do for PlatformIO.

Please note PlatformIO is full neutral organization. We don't belong to any silicon vendor.

If you need PIO Unit Testing, please mail to contact@pioplus.com and we will upgrade your account for free.

pejgstoehl, 2019-06-13:
More than half a year has passed and there is still no change. I also spent a morning getting things set up just to stumble across this issue. 

The "New in version 3.0: (PIO Plus)" disclaimer is not clear enough by any means. I had no idea PIO Plus even existed until now and I have been using PlatformIO for several years. Please just add something like "This feature requires a paid PIO plus subscription". The way it is now it feels like you are trying to hide that fact.

ivankravets, 2019-06-18:
Sorry for the delay! 

[PlatformIO Plus Goes Open Source, Improving Embedded Development Community Worldwide](https://community.platformio.org/t/platformio-plus-goes-open-source-improving-embedded-development-community-worldwide/8240/4).

closed 2019-06-18 by ivankravets

ianfixes, 2019-06-19:
Which commit (or range of commits) contains the added functionality?  Also, does this functionality require a network connection to platform.io's servers?

ivankravets, 2019-06-20:
It does not require Internet connection. See 
- https://github.com/platformio/platformio-core/tree/develop/platformio/commands/test 
- https://github.com/platformio/platformio-core/commit/948a977fa543a7f70eebaabaa5f836323f0906cf