platformio/platformio-examples #32: Hexadecimal print display problem

cgrrty, 2019-11-04:
Test code is as follows：
``void setup()
{
  Serial.begin(115200);
}

void loop()
{
  Serial.println("test...");
  uint8_t ucnt = 0;
  int8_t icnt = 0; 

  while (1)
  {
    ucnt ++;
    icnt ++;    
    Serial.printf("%x  %x\r\n",ucnt,icnt);     
  }
}
Print result
...
7a  7a
7b  7b
7c  7c
7d  7d
7e  7e
7f  7f
80  ffffff80
81  ffffff81
82  ffffff82
83  ffffff83
84  ffffff84
...
**The result should also be 80, but the result is  ffffff80.**

ivankravets, 2019-11-04:
Does it print well in Arduino IDE?

cgrrty, 2019-11-06:
The results in the Arduino IDE are the same.

cgrrty, 2019-11-06:
How to correctly print a data（signed） in hexadecimal？

ivankravets, 2019-11-06:
Please forward to https://community.platformio.org/

closed 2019-11-06 by ivankravets

cgrrty, 2019-11-06:
Please tell me how to do it in detail.