platformio/platformio-examples #34: arduino-mock does not compile with platformio 4.1.0

sanddorn, 2019-12-27:
Trying the native example for arduino-mock fails while compiling. The "Arduino.h" header is not found. It does exist in the ArduinoFake-Lib for the native target. There might be a missing -I for the gcc to find the native libs.

```
➜  arduino-mock git:(develop) ✗ pio test -v
Verbose mode can be enabled via `-v, --verbose` option
Collected 1 items

Processing * in native environment
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Building...
Verbose mode can be enabled via `-v, --verbose` option
PACKAGES: 
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <MyService>
Building in release mode
Compiling .pio/build/native/test/test_my_service.o
Compiling .pio/build/native/libd56/MyService/MyService.o
test/test_my_service.cpp:1:10: fatal error: 'Arduino.h' file not found
#include <Arduino.h>
         ^~~~~~~~~~~
In file included from lib/MyService/MyService.cpp:1:
lib/MyService/MyService.h:3:10: fatal error: 'Arduino.h' file not found
#include <Arduino.h>
         ^~~~~~~~~~~
1 error generated.
*** [.pio/build/native/libd56/MyService/MyService.o] Error 1
1 error generated.
*** [.pio/build/native/test/test_my_service.o] Error 1
========================================================================================================================== [FAILED] Took 1.24 seconds ==========================================================================================================================

Test    Environment    Status    Duration
------  -------------  --------  ------------
*       native         FAILED    00:00:01.244
===================================================================================================================== 1 failed, 0 succeeded in 00:00:01.244 =====================================================================================================================
➜  arduino-mock git:(develop) ✗ pio --version
PlatformIO, version 4.1.0

```

valeros, 2019-12-27:
Hi @sanddorn ! Could you please upload somewhere a minimal example to reproduce the issue?

sanddorn, 2019-12-28:
Hi @valeros , 

platformio provides the example, in the platformio-examples you'll find unit-testing/arduino-mock. Simply start ``pio test`` in that directory.

valeros, 2019-12-30:
Hi @sanddorn ! Could you please try adding `lib_compat_mode = off` option to your `platformio.ini` file, something like this:
```
[env:native]
platform = native
build_flags = -std=gnu++11
lib_compat_mode = off
lib_deps =
    ArduinoFake
```

sanddorn, 2019-12-31:
Hi @valeros, 

your hint works. I added a correction for the arduino-mock submodule. See https://github.com/platformio/platformio-examples/pull/35. 

Thank you for your help.

Karosuo, 2020-02-20:
Hi @valeros and @sanddorn, thank you for your contribution, quick question.
Should we close this issue since the PR #5 fixed the target of the PR #35 that fixes this one?
Just trying to start helping the community.

sanddorn, 2020-02-20:
Since ArduinoMock released the bugfix in 0.2.1 and the fix works, I'll close this issue.

closed 2020-02-20 by sanddorn