platformio/platformio-examples #38: need good first example for unit testing

MkLHX, 2020-07-17:
Hi there!

I'm trying to make unit tests during my dev. And after many time to understand something about. I'm facing of an error when i try to making tests.

I'm on platformio VSCode
core: 4.3.4
Home: 3.2.3

On windows 10 Pro
Version 1909
Os Build 18363.959

my platformio.ini
```
[env:pro16MHzatmega328]
platform = atmelavr
board = pro16MHzatmega328
framework = arduino
monitor_speed = 115200
upload_port=COM4
; test_build_project_src = yes

[env:uno]
platform = atmelavr
board = uno
framework = arduino
monitor_speed = 115200
upload_port=COM5
```

my main.h
```
#ifndef MAIN_H
#define MAIN_H
#include <Arduino.h>

#define ARRAY_SIZE(arr) int(sizeof(arr) / sizeof(arr[0]))

/**
 * @brief array of pins for pumps in1 & in2
 * 
 */
const int PUMPS_PINS[5] = {2, 3, 4, 5, 6};
const int PUMPS_PINS_TODO = {15, 16, 17, 18, 19};

#endif // MAIN_H
```
my main.cpp
```

```

my test_main.cpp
```
#include <Arduino.h>
#include <unity.h>
#include <main.h>

/**
 * @brief unit tests
 *  
 */

// void setUp(void) {
// // set stuff up here
// }

// void tearDown(void) {
// // clean stuff up here
// }

void test_pumps_pin_number(int ppId)
{
    TEST_ASSERT_EQUAL(PUMPS_PINS_TODO[ppId], PUMPS_PINS[ppId]);
}

void test_pump_state_high(int ppId)
{
    digitalWrite(PUMPS_PINS[ppId], HIGH);
    TEST_ASSERT_EQUAL(HIGH, digitalRead(PUMPS_PINS[ppId]));
}

void test_pump_state_low(int ppId)
{
    digitalWrite(PUMPS_PINS[ppId], LOW);
    TEST_ASSERT_EQUAL(LOW, digitalRead(PUMPS_PINS[ppId]));
}

uint8_t i = 0;

void setup()
{
    // NOTE!!! Wait for >2 secs
    // if board doesn't support software reset via Serial.DTR/RTS
    delay(2000);
    UNITY_BEGIN(); // IMPORTANT LINE!
    for (int i = 0; i < ARRAY_SIZE(PUMPS_PINS_TODO); i++)
    {
        RUN_TEST(test_pumps_pin_number(i));
        pinMode(PUMPS_PINS_TODO[i], OUTPUT);
    }
}

void loop()
{
    if (i < ARRAY_SIZE(PUMPS_PINS_TODO))
    {
        RUN_TEST(test_pump_state_high(i));
        delay(500);
        RUN_TEST(test_pump_state_low(i));
        delay(500);
        i++;
    }
    else if (i == ARRAY_SIZE(PUMPS_PINS_TODO))
    {
        UNITY_END(); // stop unit testing
    }
}
```
The error 
>In file included from C:\Users\mickaponik\.platformio\packages\tool-unity/unity.h:21:0,
                 from test\test_main.cpp:2:
test\test_main.cpp: In function 'void setup()':
C:\Users\mickaponik\.platformio\packages\tool-unity/unity_internals.h:707:65: error: invalid use of void expression
 #define RUN_TEST(func) UnityDefaultTestRun(func, #func, __LINE__)

So is it possible to pass an argument on test_methods() ?

valeros, 2020-07-17:
Hi @MkLHX !
> So is it possible to pass an argument on test_methods() ?

Unfortunately, it's not possible. You need to move all your test logic with `for` loops to standalone functions, e.g.

```
...
void test_pumps_pins()
{
    for (int i = 0; i < ARRAY_SIZE(PUMPS_PINS_TODO); i++)
    {
        test_pumps_pin_number(i);
        pinMode(PUMPS_PINS_TODO[i], OUTPUT);
    }
}

void setup()
{
    ....
    UNITY_BEGIN();
    RUN_TEST(test_pumps_pins);
    ...
}
...
```

Useful links:

- [Unit testing examples](https://github.com/platformio/platformio-examples/tree/develop/unit-testing) in this repository.
- [Getting started guide](https://github.com/ThrowTheSwitch/Unity/blob/master/docs/UnityGettingStartedGuide.md) for Unity framework.

closed 2020-07-17 by valeros

MkLHX, 2020-07-17:
thx!

i'd looked for examples unti-testing but do not find how to test my own code on main.cpp. 

Step by step i go forward.