platformio/platformio-examples #39: Question of debug Unit testing program.

ysong-aceinna, 2020-08-07:
Hi, I want to debug the Unit testing program 'unit-testing/calculator/', what I did is click the 'Pre-Debug' button under 'env:native', and got a feedback 'Error: Please setup environments in `platformio.ini` file'.
Could you please tell me how to setup environments in `platformio.ini' to fix this? Thank you!

[image]

ivankravets, 2020-08-07:
There is a feature request for this https://github.com/platformio/platformio-core/issues/948

We plan to implement this soon.

closed 2020-08-07 by ivankravets

lfenster, 2020-11-23:
has this been implemented yet?

ysong-aceinna, 2020-12-14:
> has this been implemented yet?

No :(