platformio/platformio-examples #41: Unittesting wont work on STM32F407 Discovery Board

mytechnotalent, 2021-08-20:
https://docs.platformio.org/en/latest/tutorials/ststm32/stm32cube_debugging_unit_testing.html

Followed the above steps exactly and updated `platform.io` as follows...
```
[env:disco_f407vg]
platform = ststm32
board = disco_f407vg
framework = stm32cube
test_transport = custom
```

Result
```
> Executing task: C:\Users\kevin\.platformio\penv\Scripts\platformio.exe test --environment disco_f407vg <

Verbose mode can be enabled via `-v, --verbose` option
Collected 1 items

Processing * in disco_f407vg environment
------------------------------------------------------------------------------------------------------------Building...
C:\Users\kevin\.platformio\packages\framework-stm32cubef4\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_smbus.c: In function 'SMBUS_MasterTransmit_TXE':
C:\Users\kevin\.platformio\packages\framework-stm32cubef4\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_smbus.c:1928:12: warning: unused variable 'CurrentMode' [-Wunused-variable]
   uint32_t CurrentMode        = hsmbus->Mode;
            ^~~~~~~~~~~
C:\Users\kevin\.platformio\packages\framework-stm32cubef4\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_smbus.c: In function 'SMBUS_Master_ADDR':
C:\Users\kevin\.platformio\packages\framework-stm32cubef4\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_smbus.c:2272:12: warning: unused variable 'CurrentXferOptions' [-Wunused-variable]
   uint32_t CurrentXferOptions = hsmbus->XferOptions;
            ^~~~~~~~~~~~~~~~~~
C:\Users\kevin\.platformio\packages\framework-stm32cubef4\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_smbus.c:2271:12: warning: unused variable 'CurrentMode' [-Wunused-variable]
   uint32_t CurrentMode        = hsmbus->Mode;
            ^~~~~~~~~~~
Uploading...
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.11.0-00155-ge392e485e (2021-03-15-16:44)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

srst_only separate srst_nogate srst_open_drain connect_deassert_srst

target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080011b0 msp: 0x20020000
** Programming Started **
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
shutdown command invoked
Testing...
If you don't see any output for the first 10 secs, please reset board (press reset button)
```

There is no reset button on this board and I can't seem to find a work around.

valeros, 2021-08-20:
Hi @mytechnotalent ! What communication channel did you use in your `unittest_transport.c`?

mytechnotalent, 2021-08-20:
@valeros 
```
#include "unittest_transport.h"
#include "stm32f4xx_hal.h"

#define USARTx                           USART2
#define USARTx_CLK_ENABLE()              __HAL_RCC_USART2_CLK_ENABLE()
#define USARTx_CLK_DISABLE()             __HAL_RCC_USART2_CLK_DISABLE()
#define USARTx_RX_GPIO_CLK_ENABLE()      __HAL_RCC_GPIOA_CLK_ENABLE()
#define USARTx_TX_GPIO_CLK_ENABLE()      __HAL_RCC_GPIOA_CLK_ENABLE()
#define USARTx_RX_GPIO_CLK_DISABLE()     __HAL_RCC_GPIOA_CLK_DISABLE()
#define USARTx_TX_GPIO_CLK_DISABLE()     __HAL_RCC_GPIOA_CLK_DISABLE()

#define USARTx_FORCE_RESET()             __HAL_RCC_USART2_FORCE_RESET()
#define USARTx_RELEASE_RESET()           __HAL_RCC_USART2_RELEASE_RESET()

#define USARTx_TX_PIN                    GPIO_PIN_2
#define USARTx_TX_GPIO_PORT              GPIOA
#define USARTx_TX_AF                     GPIO_AF7_USART2
#define USARTx_RX_PIN                    GPIO_PIN_3
#define USARTx_RX_GPIO_PORT              GPIOA
#define USARTx_RX_AF                     GPIO_AF7_USART2

static UART_HandleTypeDef UartHandle;

void unittest_uart_begin()
{
   GPIO_InitTypeDef  GPIO_InitStruct;

  USARTx_TX_GPIO_CLK_ENABLE();
  USARTx_RX_GPIO_CLK_ENABLE();

  USARTx_CLK_ENABLE();

  GPIO_InitStruct.Pin       = USARTx_TX_PIN;
  GPIO_InitStruct.Mode      = GPIO_MODE_AF_PP;
  GPIO_InitStruct.Pull      = GPIO_PULLUP;
  GPIO_InitStruct.Speed     = GPIO_SPEED_FAST;
  GPIO_InitStruct.Alternate = USARTx_TX_AF;

  HAL_GPIO_Init(USARTx_TX_GPIO_PORT, &GPIO_InitStruct);

  GPIO_InitStruct.Pin = USARTx_RX_PIN;
  GPIO_InitStruct.Alternate = USARTx_RX_AF;

  HAL_GPIO_Init(USARTx_RX_GPIO_PORT, &GPIO_InitStruct);
  UartHandle.Instance          = USARTx;

  UartHandle.Init.BaudRate     = 115200;
  UartHandle.Init.WordLength   = UART_WORDLENGTH_8B;
  UartHandle.Init.StopBits     = UART_STOPBITS_1;
  UartHandle.Init.Parity       = UART_PARITY_NONE;
  UartHandle.Init.HwFlowCtl    = UART_HWCONTROL_NONE;
  UartHandle.Init.Mode         = UART_MODE_TX_RX;
  UartHandle.Init.OverSampling = UART_OVERSAMPLING_16;

  if(HAL_UART_Init(&UartHandle) != HAL_OK) {
    while(1){}
  }

}

void unittest_uart_putchar(char c)
{
    HAL_UART_Transmit(&UartHandle, (uint8_t*)(&c), 1, 1000);
}

void unittest_uart_flush(){}

void unittest_uart_end() {
  USARTx_CLK_DISABLE();
  USARTx_RX_GPIO_CLK_DISABLE();
  USARTx_TX_GPIO_CLK_DISABLE();
}
```
Not clear exactly on what channel so I pasted everything.

mytechnotalent, 2021-08-20:
@valeros 
`unittest_transport.h` is saying include file not found in browse path.  

test
    test_main.c
    unittest_transport.c
    unittest_transport.h

That is where it is placed is it in the wrong place?

mytechnotalent, 2021-08-20:
Even with `unittest_transport.h' put in `include` dir now the error goes away but same result.

```
> Executing task: C:\Users\kevin\.platformio\penv\Scripts\platformio.exe test --environment disco_f407vg <

Verbose mode can be enabled via `-v, --verbose` option
Collected 1 items

Processing * in disco_f407vg environment
------------------------------------------------------------------------------------------------------------Building...
C:\Users\kevin\.platformio\packages\framework-stm32cubef4\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_smbus.c: In function 'SMBUS_MasterTransmit_TXE':
C:\Users\kevin\.platformio\packages\framework-stm32cubef4\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_smbus.c:1928:12: warning: unused variable 'CurrentMode' [-Wunused-variable]
   uint32_t CurrentMode        = hsmbus->Mode;
            ^~~~~~~~~~~
C:\Users\kevin\.platformio\packages\framework-stm32cubef4\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_smbus.c: In function 'SMBUS_Master_ADDR':
C:\Users\kevin\.platformio\packages\framework-stm32cubef4\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_smbus.c:2272:12: warning: unused variable 'CurrentXferOptions' [-Wunused-variable]
   uint32_t CurrentXferOptions = hsmbus->XferOptions;
            ^~~~~~~~~~~~~~~~~~
C:\Users\kevin\.platformio\packages\framework-stm32cubef4\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_smbus.c:2271:12: warning: unused variable 'CurrentMode' [-Wunused-variable]
   uint32_t CurrentMode        = hsmbus->Mode;
            ^~~~~~~~~~~
Uploading...
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.11.0-00155-ge392e485e (2021-03-15-16:44)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

srst_only separate srst_nogate srst_open_drain connect_deassert_srst

target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080003bc msp: 0x20020000
** Programming Started **
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
shutdown command invoked
Testing...
If you don't see any output for the first 10 secs, please reset board (press reset button)
```
Hangs and does not do the unittests.

valeros, 2021-08-20:
That tutorial was written for Nucleo-F401RE which has a completely different schematic, so it won't work with your board, mainly because the ST-LINK VCP on the Disco-F407VG board is not even connected to the MCU. I see here two options:
- Solder the ST-LINK VCP pins to any available UART pins on your board and rewrite the `unittest_transport.c` to use that UART.
- Use an external USB-UART converter, connect it to any available UART pins and rewrite the `unittest_transport.c` to use that UART.

closed 2021-08-20 by mytechnotalent